opensocsysarch/CoreGen #201: Ubuntu Xenial 16.04 Virtual Machine Build Failure

peteherman, 2019-11-26:
When attempting to build from source I encountered two errors while following the build steps outlined in the README.md.

1.) Overriding the default compiler.
     - When overriding the default compiler I had to specify the version of clang which existed on my machine.
     - ex: `export CC=clang-6.0`
     - ex: `export CXX=clang++-6.0`

2.) Linker searching for libraries with a different name.
     - When executing the `build` I encountered a failure when CoreGen was building the Sample Plugin. The failure message stated that "libpyclingo.so could not be found."  An `strace` revealed the gcc linker was looking for a library with the name "liblibpyclingo.so" rather than libplycingo.so. 
     - This issue was resolved by creating a symbolic link to the correctly named library.
     - The same issue occurred again later in the build, this time with the libz.so library. No libz.so library comes in box in Xenial 16.04, but rather this library is named libz.so.1. 
     - Again this issue was resolved by creating a symbolic link to the correctly named library.

jleidel, 2019-11-26:
@peteherman, thanks for submitting the issue.  We'll reproduce this locally and attempt to get a patch merged in asap.

jleidel, 2019-11-26:
Peter, on 1) this will be platform dependent.  The default llvm-6.0 installation does not put Clang/Clang++ in the default path.  In which case, you can use `export CC=/usr/lib/llvm-6.0/bin/clang`; or `export CC=/usr/lib/llvm-6.0/bin/clang-6.0`.  

On 2) where are you building the source from?  Is this done out of tree (eg, a 'build' directory)?  

I'm attempting reproduce your environment

jleidel, 2019-11-27:
I've been able to reproduce the issue on a raw VM system; working the issue

jleidel, 2019-11-29:
merge into devel complete; waiting for CI tests to run+pass

jleidel, 2019-11-29:
passed all devel tests; merging to master on next major merge cycle

closed 2019-11-29 by jleidel