opensocsysarch/CoreGen #303: Add constant routing/wire congestion overhead calculation metrics

jleidel, 2021-01-04:
Add constant routing/wire congestion overhead calculation metrics

jleidel, 2021-01-04:
This feature should allow users to specify a notional overhead amount on the command line.  This will be added to the target values when deriving the output metrics.  For example: 

`poar --design example.yaml --overhead 12`

...will add 12% routing/congestion overhead to all the values

rkabrick, 2021-01-15:
I feel like this could be generalized such that there is a vector of multipliers which can be cycled through and applied during the data derivation stage (while keeping the vanilla value produced by the accumulator) so we don't limit ourselves to just this one metric in the future

jleidel, 2021-01-15:
That will work, feel free to implement that

closed 2021-01-18 by rkabrick