opensocsysarch/CoreGen #338: Coren Error ：No return parameters

dingzex, 2021-07-28:
Hello!
When testing and using Coren, when "cgcli -- IR basicrisc.yaml -- compiler" is made for example level1 / step10, the "error executing the llvm CodeGen:" is output. What is the reason?

jleidel, 2021-07-28:
You need to be more specific.  What build options did you use, what where the EXACT command line options

dingzex, 2021-07-28:
I did the test completely according to the tutorial:
cgcli Parameter reference help:
 Execution Options:
	--verify                              : Verify the IR and exit
	--pass                                : Run some or all of the IR passes
	--chisel                              : Run the Chisel codegen
	--compiler [Optional Version]         : Run the compiler codegen

ding@ding-Asip:~/SystemArchitect/CoreGenTutorials/LEVEL1/Step10$ cgcli -ir BasicRISC.yaml --verify
ding@ding-Asip:~/SystemArchitect/CoreGenTutorials/LEVEL1/Step10$ cgcli -ir BasicRISC.yaml --pass
ding@ding-Asip:~/SystemArchitect/CoreGenTutorials/LEVEL1/Step10$ cgcli -ir BasicRISC.yaml --chisel
ding@ding-Asip:~/SystemArchitect/CoreGenTutorials/LEVEL1/Step10$ cgcli -ir BasicRISC.yaml --compiler

Error executing the LLVM codegen: 
[image]

jleidel, 2021-07-28:
Did you initialize the System Architect archive first?

dingzex, 2021-07-28:
Yes, I have run:
cgcli --init archive/master.yaml

dingzex, 2021-07-28:
After I create html and man directories in Doxygen directory, it can be executed normally:

sudo make install

Then, perform the test tutorial:

Execute cgcli -- ir Basicrisc.yaml -- compiler

appear:

Error executing the LLVM codegen:

Without any prompt, the test process is terminated.

dingzex, 2021-07-28:
1

closed 2021-07-30 by jleidel

dingzex, 2021-08-01:
I wonder if you will continue to improve this project?