opensocsysarch/CoreGen #339: CMake Error

dingzex, 2021-07-28:
I'm trying to compile coregen, but the following problems occur during installation. Make install error:
coregen@coregen-desktop:~/System_Architect/CoreGen/build$ sudo make install
[ 15%] Built target yaml-cpp
[ 16%] Built target DataPassObjs
[ 24%] Built target AnalysisPassObjs
[ 25%] Built target OptPassObjs
[ 28%] Built target NodeCodegenObjs
[ 29%] Built target LLVMCodegenObjs
[ 29%] Built target SMPassObjs
[ 31%] Built target CoreGenSigMap
[ 32%] Built target CoreGenArchive
[ 35%] Built target CoreGenCodegen
[ 36%] Built target CoreGenPluginImpl
[ 49%] Built target CoreGenBackend
[ 50%] Built target SamplePlugin
[ 64%] Built target IntrinObjs
[ 67%] Built target PassObjs
[ 72%] Built target SCComp
[ 73%] Built target sccomp
[ 74%] Built target scsigv
[ 76%] Built target cgcli
[ 77%] Built target gmock
[ 83%] Built target run-tests
[ 85%] Built target gmock_main
[ 86%] Built target gtest
[ 87%] Built target gtest_main
[ 88%] Built target parse
[ 89%] Built target read
[ 90%] Built target sandbox
[ 91%] Built target coregen-config
[ 95%] Built target AccumObjs
[ 98%] Built target PoarBackend
[100%] Built target poar
Install the project...
-- Install configuration: ""
CMake Error at cmake_install.cmake:36 (file):
  file INSTALL cannot find
  "/home/coregen/System_Architect/CoreGen/doxygen/html".

make: *** [Makefile:95：install] 错误 1
coregen@coregen-desktop:~/System_Architect/CoreGen/build$

jleidel, 2021-07-28:
Installing from source requires that you build the Doxygen documentation. Do the following: 
- Add `-DBUILD_DOCUMENTATION=ON` to the CMake options
- Rerun the build
- Build the docs using `make doc`
- Install everything: `sudo make install`

closed 2021-07-28 by jleidel

dingzex, 2021-07-28:
According to the parameters you provide, the same result will be obtained after recompilation
[image]

jleidel, 2021-07-28:
Did you remove the entire build directory first?  Any time you rerun cmake, you need to remove the build prior to rerunning cmake.  You also need to ensure that you have Doxygen installed.

dingzex, 2021-07-28:
I reinstalled the project, but there are still installation errors:
1.Override default compiler
export CC=/usr/lib/llvm-8/bin/clang
export CXX=/usr/lib/llvm-8/bin/clang++
2.Execute CMake to generate the makefiles
ding@ding-desktop:~/CoreGen/build$ cmake -DBUILD_DOCUMENTATION=ON -DLLVM_DIR=/usr/lib/llvm-8/cmake ../
-- Building for Linux
-- COREGEN_LINK_LIBS set to dl
-- Found LLVM 8.0.1
-- Using LLVMConfig.cmake in: /usr/lib/llvm-8/cmake
-- COREGEN_INCLUDE_PATH set to /home/ding/CoreGen/include
-- STONECUTTER_PATH set to /home/ding/CoreGen/src/StoneCutter
-- COREGEN_PATH set to /home/ding/CoreGen/src/CoreGenBackend
-- COREGEN_PLUGIN_PATH set to /home/ding/CoreGen/src/CoreGenPlugin
-- COREGEN_CODEGEN_PATH set to /home/ding/CoreGen/src/CoreGenCodegen
-- COREGEN_BUILD_DIR set to /home/ding/CoreGen/build
-- YAML_PATH set to /home/ding/CoreGen/src/thirdparty/yaml/yaml-cpp-0.6.2
-- llvm_libs set to LLVMSupport;LLVMCore;LLVMIRReader;LLVMMCJIT;LLVMipo;LLVMExecutionEngine;LLVMInstrumentation;LLVMTransformUtils;LLVMVectorize;LLVMScalarOpts;LLVMInstCombine;LLVMBitReader;LLVMBitWriter;LLVMX86CodeGen;LLVMX86AsmParser;LLVMX86AsmPrinter;LLVMX86Desc;LLVMX86Disassembler;LLVMX86Info;LLVMX86Utils
-- FOUND BASH AT /usr/bin/bash
-- COREGEN_TEST_PLUGIN set to libSamplePlugin.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ding/CoreGen/build

3.make

4.sudo make install

5.Error information
....
....
[ 89%] Built target read
[ 90%] Built target sandbox
[ 91%] Built target coregen-config
[ 95%] Built target AccumObjs
[ 98%] Built target PoarBackend
[100%] Built target poar
Install the project...
-- Install configuration: ""
CMake Error at cmake_install.cmake:36 (file):
  file INSTALL cannot find "/home/ding/CoreGen/doxygen/html".

make: *** [Makefile:95：install] error 1

jleidel, 2021-07-28:
As I stated above, you have to run `make doc` before running `make install`.  `make doc` builds all the documentation using Doxygen